Freetz-NG/freetz-ng #743: Test-Patche für 5590-Labor

maverick3005, 2023-02-22:
Anbei 2 dirty vorbereitete Patches für die 5590-Labor

[0001-Adding-5590-Labor.patch](https://github.com/Freetz-NG/freetz-ng/files/10808858/0001-Adding-5590-Labor.patch)
[5590--07_5X.in.patch](https://github.com/Freetz-NG/freetz-ng/files/10808860/5590--07_5X.in.patch)

srry. Auf die Schnelle hab ich den 2. Patch irgendwie als neu erstelltes File nicht mit in den github-Patch reinbekommen.
Ich habe es daher als konventionellen Patch mal erzeugt und dazu gepackt.

Jobbedingt konnte ich auf die Schnelle jetzt nicht mehr machen, als das mal eben zusammenschrauben mit den mir bekannten
Infos.

@fda77 

Könntest Du die Basisdaten zumindest einpflegen und dann dieses für mich bömische "autogenerate" git-zeuchs noch drüber laufen lassen.... Ich konnte das mangels "generate" kram lokal alles nicht testen.

Zusatzinfos:

Der Kernel ist 4.4.271

Die mir bekannten Download urls lauten:

https://avm.de/fileadmin/user_upload/DE/Labor/Download/fritzbox-5590-fiber-labor-102864.zip

sowie den Imagefilenamen: "FRITZ.Box_5590_Fiber-07.39-102864-LabBETA.image"
in diesem ZIP-File.

ansonsten für die Inhaus:
https://download.avm.de/inhaus/MOVE21/5590Fiber/FRITZ.Box_5590_Fiber-07.39-102064-Inhaus.image
Eine Labor-URL habe ich im Juis irgendwie nicht finden können.

Zur Image-Info:
Es sind jetzt 2 extra binary-Sachen noch im Image-install-script: a.) neues Urloaderupdate und b.) tzupdate für die "Trustzone".
Die Trustzone wurde bei der Version FOS 7.30 noch nicht upgedated.

fda77, 2023-02-22:
Schön dass du es dir mal angeschaut hast. Der 2. Patch ist überflüssig, das wird gleich automatisch erstellt.
Daher ist es egal welcher kernel es ist etc. Falls die alten Sourcen nicht mehr passen wird man das erst beim make sehen, daher hab ich die config/mod/source.in weggelassen
Die "labor"  von 7490 und 5590 funktionieren scheinbar nicht via juis, avm halt...
Es gibt eine neuere inhaus: https://freetz-ng.github.io/freetz-ng/juis/#fos-inhaus , würd ich aber nicht installieren
Was avm da noch so reingepackt hat ist egal, das bleibt einfach wie es ist

closed 2023-02-22 by fda77

maverick3005, 2023-02-22:
Ah erst nach dem Ticketclose gelesen gehabt ;-)

Ja was ich so mitbekommen habe gibts mit den vielen neuen Klein-FTTH Anbietern überall in Europa wohl noch massiv
GPON Gegenstellen-Probleme, ferner verlieren viele User nach kurzer Zeit Portfreigaben bei ipv6, oder wenn WLAN mit genutzt wird crashed die Box bei der FPS 7.30 in Verbindung mit 5 GHz noch recht häufig. Das soll mit der Labor wohl laut erster Berichte jetzt besser geworden sein...

Das muss dann aber mein Bekannter selber raus finden, ich compile Ihm das Ding nur. Hin und Her wechseln kann er selber.