Freetz-NG/freetz-ng #764: Can't build menuconfig

alexceltare2, 2023-03-13:
The following occurs:
`building ...   HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/confdata.o
  LEX     scripts/kconfig/lexer.lex.c
  HOSTCC  scripts/kconfig/expr.o
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTCC  scripts/kconfig/mconf.o
  HOSTCC  scripts/kconfig/lxdialog/checklist.o
  HOSTCC  scripts/kconfig/lxdialog/inputbox.o
  HOSTCC  scripts/kconfig/lxdialog/menubox.o
  HOSTCC  scripts/kconfig/lxdialog/textbox.o
  HOSTCC  scripts/kconfig/lxdialog/yesno.o
  HOSTCC  scripts/kconfig/lxdialog/util.o
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTLD  scripts/kconfig/mconf
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-redhat-linux/11/../../../libncursesw.so.6 when searching for libncursesw.so.6
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-redhat-linux/11/../../../libtinfo.so when searching for -ltinfo
/usr/bin/ld: skipping incompatible /lib/libtinfo.so when searching for -ltinfo
/usr/bin/ld: skipping incompatible /usr/lib/libtinfo.so when searching for -ltinfo
/usr/bin/ld: cannot find -ltinfo
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-redhat-linux/11/../../../libtinfo.so when searching for -ltinfo
/usr/bin/ld: skipping incompatible /lib/libtinfo.so when searching for -ltinfo
/usr/bin/ld: skipping incompatible /usr/lib/libtinfo.so when searching for -ltinfo
collect2: error: ld returned 1 exit status
make[3]: *** [/home/alex/freetz-ng/source/host-tools/kconfig-v6.2/scripts/Makefile.host:120: scripts/kconfig/mconf] Error 1
make[2]: *** [Makefile:78: menuconfig] Error 2
make[1]: *** [make/host-tools/kconfig-host/kconfig-host.mk:31: /home/alex/freetz-ng/source/host-tools/kconfig-v6.2/scripts/kconfig/mconf] Terminated
make: *** [Makefile:46: envira] Terminated
`

AkteX, 2023-03-14:
Irgendwie scheinen die Redhat Linux Libraries inkompatibel zu sein. Ubuntu wäre evtl. einen Versuch wert.

fda77, 2023-03-18:
There where changes some weeks ago.
Please read https://github.com/Freetz-NG/freetz-ng/issues/627 and 99f551e509561468e2ae3c499d1b6867224117ac + f42f5d47a254b258f50be55eed5f3039d1ed75ff

alexceltare2, 2023-03-18:
Using Fedora 35